ThisIsNotRocketScience/GerberTools #147: Panilizer partially rotates footprint

tommy1394, 2023-01-02:
i have a design with a LFPAK33 footprint FET.  I am trying to panelize a 2x5 arraingment of a small PCB which uses this FET.  Gerber panelizer partially rotates the footprint for this LFPAK33 part.  Has anyone experienced this or have any suggestions

[image]

CRImier, 2023-01-02:
could you upload your gerber zip and .gerberset project file?

tommy1394, 2023-01-02:
yes, let me put together a zip packager...do you want the saved panelizer configuration as well?

CRImier, 2023-01-02:
yea, that's the `.gerberset`. I'm not the dev, but, this is what'd be useful for devs for triage

tommy1394, 2023-01-02:
here are the files
[panelizer-rotate-issue.zip](https://github.com/ThisIsNotRocketScience/GerberTools/files/10333358/panelizer-rotate-issue.zip)

tommy1394, 2023-01-02:
on a side note, this takes more than 8 hours to process... is this normal?

StijnKuipers, 2023-01-03:
are you building from source or using a downloaded binary? What version are you on?

A lot of macro stuff has been fixed - but not in a released bin yet.

and yes please - upload the gerbers somewhere. 

8 hours sounds extremely odd