ThisIsNotRocketScience/GerberTools #149: Cannot build on Manjaro Linux

ShadeTechnik, 2023-08-21:
Probably a version/dependency thing but I have no idea.....

`
$ ./build.sh                                                                                                                                                           
+ export TERM=xterm
+ TERM=xterm
+ nuget Triangle 0.0.6-Beta3
+ package=Triangle
+ version=0.0.6-Beta3
++ mktemp
+ zip=/tmp/tmp.XazLZRoLIi.zip
+ url=https://www.nuget.org/api/v2/package/Triangle/0.0.6-Beta3
+ dir=GerberProjects/packages/Triangle.0.0.6-Beta3
+ test -d GerberProjects/packages/Triangle.0.0.6-Beta3
+ wget https://www.nuget.org/api/v2/package/Triangle/0.0.6-Beta3 -O /tmp/tmp.XazLZRoLIi.zip
--2023-08-21 08:52:00--  https://www.nuget.org/api/v2/package/Triangle/0.0.6-Beta3
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving www.nuget.org (www.nuget.org)... 52.240.159.111
Connecting to www.nuget.org (www.nuget.org)|52.240.159.111|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://globalcdn.nuget.org/packages/triangle.0.0.6-beta3.nupkg [following]
--2023-08-21 08:52:01--  https://globalcdn.nuget.org/packages/triangle.0.0.6-beta3.nupkg
Resolving globalcdn.nuget.org (globalcdn.nuget.org)... 152.199.4.184
Connecting to globalcdn.nuget.org (globalcdn.nuget.org)|152.199.4.184|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 231470 (226K) [application/octet-stream]
Saving to: ‘/tmp/tmp.XazLZRoLIi.zip’

/tmp/tmp.XazLZRoLIi.zip                              100%[=====================================================================================================================>] 226.04K  --.-KB/s    in 0.03s   

2023-08-21 08:52:01 (7.26 MB/s) - ‘/tmp/tmp.XazLZRoLIi.zip’ saved [231470/231470]

+ mkdir -p GerberProjects/packages/Triangle.0.0.6-Beta3
+ unzip /tmp/tmp.XazLZRoLIi.zip -d GerberProjects/packages/Triangle.0.0.6-Beta3
Archive:  /tmp/tmp.XazLZRoLIi.zip
  inflating: GerberProjects/packages/Triangle.0.0.6-Beta3/_rels/.rels  
  inflating: GerberProjects/packages/Triangle.0.0.6-Beta3/Triangle.nuspec  
  inflating: GerberProjects/packages/Triangle.0.0.6-Beta3/lib/net35/Triangle.dll  
  inflating: GerberProjects/packages/Triangle.0.0.6-Beta3/lib/net45/Triangle.dll  
  inflating: GerberProjects/packages/Triangle.0.0.6-Beta3/lib/netstandard1.5/Triangle.dll  
  inflating: GerberProjects/packages/Triangle.0.0.6-Beta3/[Content_Types].xml  
 extracting: GerberProjects/packages/Triangle.0.0.6-Beta3/.signature.p7s  
+ rm /tmp/tmp.XazLZRoLIi.zip
+ nuget DotNetZip 1.12.0
+ package=DotNetZip
+ version=1.12.0
++ mktemp
+ zip=/tmp/tmp.7NevDa2kF0.zip
+ url=https://www.nuget.org/api/v2/package/DotNetZip/1.12.0
+ dir=GerberProjects/packages/DotNetZip.1.12.0
+ test -d GerberProjects/packages/DotNetZip.1.12.0
+ wget https://www.nuget.org/api/v2/package/DotNetZip/1.12.0 -O /tmp/tmp.7NevDa2kF0.zip
--2023-08-21 08:52:01--  https://www.nuget.org/api/v2/package/DotNetZip/1.12.0
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving www.nuget.org (www.nuget.org)... 52.240.159.111
Connecting to www.nuget.org (www.nuget.org)|52.240.159.111|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://globalcdn.nuget.org/packages/dotnetzip.1.12.0.nupkg [following]
--2023-08-21 08:52:01--  https://globalcdn.nuget.org/packages/dotnetzip.1.12.0.nupkg
Resolving globalcdn.nuget.org (globalcdn.nuget.org)... 152.199.4.184
Connecting to globalcdn.nuget.org (globalcdn.nuget.org)|152.199.4.184|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 444408 (434K) [application/octet-stream]
Saving to: ‘/tmp/tmp.7NevDa2kF0.zip’

/tmp/tmp.7NevDa2kF0.zip                              100%[=====================================================================================================================>] 433.99K  --.-KB/s    in 0.05s   

2023-08-21 08:52:02 (8.65 MB/s) - ‘/tmp/tmp.7NevDa2kF0.zip’ saved [444408/444408]

+ mkdir -p GerberProjects/packages/DotNetZip.1.12.0
[... 432 lines cut ...]

Errors:

/home/justin/git/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->
/home/justin/git/GerberTools/GerberLibrary/GerberLibrary.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.CSharp.targets (CoreCompile target) ->

	Core/Gerber.cs(14,7): error CS0246: The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)
	Artwork Related/GerberOutlineWriter.cs(4,7): error CS0246: The type or namespace name 'QiHe' could not be found (are you missing a using directive or an assembly reference?)
	Core/GerberPanel.cs(13,7): error CS0246: The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)
	Core/ImageCreator.cs(13,7): error CS0246: The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)
	Core/SickOfBeige.cs(13,7): error CS0246: The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)

/home/justin/git/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->

	/home/justin/git/GerberTools/GerberProjects/KicadPartsToCSV/KicadPartsToCSV.csproj: error : /home/justin/git/GerberTools/GerberProjects/KicadPartsToCSV/KicadPartsToCSV.csproj: The default XML namespace of the project must be the MSBuild XML namespace. If the project is authored in the MSBuild 2003 format, please add xmlns="http://schemas.microsoft.com/developer/msbuild/2003" to the <Project> element. If the project has been authored in the old 1.0 or 1.2 format, please convert it to MSBuild 2003 format.  

	 10 Warning(s)
	 6 Error(s)

Time Elapsed 00:00:03.6874240

`

whitelynx, 2023-09-05:
I'm having the same issue. I was able to get rid of some of the issues by making a couple of changes to include `DotNetZip` (which apparently is the new drop-in replacement for `Ionic.Zip`) but I have no idea how to find `QiHe.CodeLib`.

Here's the changes I made:
```diff
diff --git a/GerberLibrary/GerberLibrary.csproj b/GerberLibrary/GerberLibrary.csproj
index c885beb..57ae871 100644
--- a/GerberLibrary/GerberLibrary.csproj
+++ b/GerberLibrary/GerberLibrary.csproj
@@ -89,6 +89,9 @@
     <Reference Include="Triangle, Version=0.0.6.0, Culture=neutral, processorArchitecture=MSIL">
       <HintPath>..\GerberProjects\packages\Triangle.0.0.6-Beta3\lib\net45\Triangle.dll</HintPath>
     </Reference>
+    <Reference Include="DotNetZip, Version=1.16.0.0, Culture=neutral, processorArchitecture=MSIL">
+      <HintPath>..\GerberProjects\packages\DotNetZip.1.16.0\lib\net40\DotNetZip.dll</HintPath>
+    </Reference>
   </ItemGroup>
   <ItemGroup>
     <Compile Include="Artwork Related\BasicBounce.cs" />
diff --git a/build.sh b/build.sh
index 23cca53..ea5388a 100755
--- a/build.sh
+++ b/build.sh
@@ -30,6 +30,7 @@ nuget netDXF 0.9.3
 nuget GlmNet 0.5.1
 nuget DockPanelSuite 3.0.6
 nuget DockPanelSuite.ThemeVS2015 3.0.6
+nuget DotNetZip 1.16.0
 
 
 xbuild /p:Configuration=Debug GerberProjects/GerberProjects.sln
```

Even after these changes, we still get the `QiHe.CodeLib` error, as well as one that makes me think that maybe we're building on too new a version of .NET:

```
/home/whitelynx/Development/Other/GerberTools/GerberProjects/GerberProjects.sln (default targets) ->
(Build target) ->

	/home/whitelynx/Development/Other/GerberTools/GerberProjects/KicadPartsToCSV/KicadPartsToCSV.csproj: error : /home/whitelynx/Development/Other/GerberTools/GerberProjects/KicadPartsToCSV/KicadPartsToCSV.csproj: The default XML namespace of the project must be the MSBuild XML namespace. If the project is authored in the MSBuild 2003 format, please add xmlns="http://schemas.microsoft.com/developer/msbuild/2003" to the <Project> element. If the project has been authored in the old 1.0 or 1.2 format, please convert it to MSBuild 2003 format.
```

----

Does anyone know where `QiHe.CodeLib` comes from? I don't see it on nuget.org or anywhere else that I've looked so far...

oliv3r, 2024-10-04:
@whitelynx looks like you are 3 steps further then what I'm getting :p I'm trying to kick this horse via docker, so that we could eventually have a reproduceable static build environment. We should be able to run this from docker too, but that's for later to worry about.

So for now, I'm doing something stupidly simple as
```sh
$ docker run --rm -it -v ./:/src -w /src debian:stable-slim /bin/bash
# DEBIAN_NON_INTERACTIVE=true apt-get update && apt-get install --yes  mono-devel wget unzip
./build.sh
```

But as everybody, shit is failing left and right. Interestingly someone added the DotNetZip dependency to build.sh ages ago, albeit with different versions, and those where bumped as well, but that project change I didn't see.

Regardless, the next thing it fails for me now is 
```
	Core/GerberPanel.cs(14,7): error CS0246: The type or namespace name `TriangleNet' could not be found. Are you missing an assembly reference?
```
which is odd, because I thought 'triangle' is resolving 'trianglenet'.

This build env being to new argument is also interesting, as of course it isn't specified anywhere what we do actually need, rolling back with docker to older stuff is trivial; but then we'd need to know of course which version ...