ThisIsNotRocketScience/GerberTools #154: Default Gerber format generated by KiCad 8 is not reconized

alzomor, 2024-08-03:
I tried to use default generated Gerbers from Ki Cad 8, but I get only output panel with drills and no other information.
I need to switch to old Protel format in order to get all layers recognized by Gereber Panelizer!

CRImier, 2024-08-05:
oh yeah, I remember doing that too. It should have support for interpreting the (iirc) `.gbr` new format, but if you're not using the latest revision, it might not be as functional.

Witawat, 2024-12-08:
> I tried to use default generated Gerbers from Ki Cad 8, but I get only output panel with drills and no other information. I need to switch to old Protel format in order to get all layers recognized by Gereber Panelizer!

i try old Protel format not work

CRImier, 2025-03-12:
@Witawat considered compiling the latest version of Gerber Panelizer? It's pretty simple to do, can recommend, and in my experience, it fixes most compatibility issues with KiCad.

Witawat, 2025-03-14:
> [@Witawat](https://github.com/Witawat) considered compiling the latest version of Gerber Panelizer? It's pretty simple to do, can recommend, and in my experience, it fixes most compatibility issues with KiCad.

can share ?